Open-CMSIS-Pack/CMSIS-Driver_STM32 #14: MX_DEVICE_VERSION is missing

MaxFlatline, 2024-10-10:
Helllo. I tried to use this pack with L0 and F4 DFPs and CUBE MX output and result was this error. All packs are latest, Keil and CubeMX too. MX_Device.h doesn't have this macros

MiloradCvjetkovic, 2024-10-14:
Hi @MaxFlatline , 
the new concept depends on support in the DFP and also latest cmsis-toolbox.
Can you try with https://github.com/Open-CMSIS-Pack/STM32F4xx_DFP/releases/tag/v3.0.0 and and https://github.com/Open-CMSIS-Pack/cmsis-toolbox/releases/tag/2.6.1
New DFP is not jet available for the L0 series.

MiloradCvjetkovic, 2024-11-07:
Can this issue be closed @MaxFlatline ?

MaxFlatline, 2024-11-07:
If the pack will be changed not to replace working solutions, then yes. Right now it acts as if it supports every single ST MCU and disrupts my workflow

HansSchneebauer, 2025-03-31:
The STM32L0xx_DFP was updated to version 3.0.0 this month which fully supports CubeMX now. See:
https://github.com/Open-CMSIS-Pack/STM32L0xx_DFP/releases/tag/v3.0.0
So I assume that this issue can be closed.

closed 2025-03-31 by HansSchneebauer