Open-CMSIS-Pack/CMSIS-Driver_STM32 #17: Communication failed with device without OTG support

minicarpet, 2024-10-18:
Hi,

I am using the board [B-L462E-CELL1](https://www.st.com/en/evaluation-tools/b-l462e-cell1.html) as sample.
I configured the device to use this new driver. Transferring the data from device to the host is ok, but from host to device data are **incorrect** 

to reproduce : create keil project using CDC configuration (see pictures) :
[image]
[image]
[image]
[image]
[image]

Starts USB:

`
/* USER CODE BEGIN Includes */
#include "rtx_os.h"                     // ARM::CMSIS:RTOS2:Keil RTX5
#include "rl_usb.h"                     // USB:CORE&&Release
/* USER CODE END Includes */

...

/* USER CODE BEGIN 0 */
uint32_t HAL_GetTick(void)
{
	uint32_t currentTick;

	if (osKernelGetState() == osKernelRunning)
	{
		currentTick = osKernelGetTickCount();
	}
	else
	{
		uint32_t i;

		/* If Kernel is not running wait approximately 1 ms then increment 
		and return auxiliary tick counter value */
		for (i = (SystemCoreClock >> 14U); i > 0U; i--)
		{
			__NOP(); __NOP(); __NOP(); __NOP(); __NOP(); __NOP();
		}
		
		HAL_IncTick();
		currentTick = uwTick;
	}

	return currentTick;
}
/* USER CODE END 0 */

void app_main (void *argument)
{
  USBD_Initialize(0);
  USBD_Connect(0);
  
  while(1)
  {
    (void)osThreadFlagsWait(0, osFlagsWaitAny, osWaitForever);
  }
}

int main(void)
{
    /* USER CODE BEGIN 2 */
    osStatus_t statusOS = osError;
  
    /* Initialize CMSIS-RTOS2 */
    statusOS = osKernelInitialize();
    if(osOK == statusOS)
    {
	/* Create application main thead */
	osThreadNew(app_main, NULL, NULL);    // Create application main thread
        osKernelStart();
    }
    /* USER CODE END 2 */
}
`

implement USBD_User_CDC_ACM_0.c as follow:

`
// Local Variables
static   CDC_LINE_CODING        cdc_acm_line_coding = { 115200U, 0U, 0U, 8U };
static   uint8_t                uart_tx_buf[64];
static   uint8_t                uart_rx_buf[64];

void USBD_CDC0_ACM_Initialize (void) {
  // Add code for initialization
  (void)memset(uart_tx_buf, 0x00, sizeof(uart_tx_buf));
  (void)memset(uart_tx_buf, 0x00, sizeof(uart_rx_buf));
}

void USBD_CDC0_ACM_DataReceived (uint32_t len) {
  // Add code for handling new data reception
  int32_t cnt = USBD_CDC_ACM_ReadData(0, uart_rx_buf, len);
  if (cnt > 0)
  {
    (void) memcpy(uart_tx_buf, "Data received : ", strlen("Data received : "));
    (void) memcpy(uart_tx_buf + strlen("Data received : "), uart_rx_buf, cnt);
    USBD_CDC_ACM_WriteData(0U, uart_tx_buf, strlen("Data received : ") + cnt);
  }
}
`

using terra term, communicate with the board, and see that's not correct return.

MiloradCvjetkovic, 2024-10-18:
In Initialize function uart_tx_buf is cleared twice instead of uart_tx_buf  and uart_rx_buf . but that should not matter for the problem.
Can you write some contents of uart_rx_buf  and uart_tx_buf , and what is different than expected?

minicarpet, 2024-10-21:
Hi
Indeed, I fixed the memory initialization.
well, using TeraTem application, after a connection.
[image]
If I press 'a', I successfully got into `USBD_CDC0_ACM_DataReceived` method. But I receive the wrong data (0x03 received instead of 0x61 ascii code for 'a'.
[image]

Then, I sent back the received data, and that is working:
[image]
[image]

But I should have somthing like this:
[image]

ReinhardKeil, 2024-10-22:
It looks like you have a test project.  Could you ZIP this project and attach the file?

minicarpet, 2024-10-22:
Sure ! Here we go:
[Uploading test_release_driver_l4.zip…]()

MiloradCvjetkovic, 2024-10-22:
Hi @minicarpet ,
something went wrong with you attaching of file, it is not accessible, can you try attaching .zip file again?

I see that you do not have CubeMX generated main function but have your main function.
The new concept of integration with CubeMX requires that CubeMX generated code is used for device initialization.

minicarpet, 2024-10-22:
Hi,
could you try again with this file ?
[test_release_driver_l4.zip](https://github.com/user-attachments/files/17472945/test_release_driver_l4.zip)

also, I do not use custom main, I just have main_thread from the CMSIS RTX started from main generated by CubeMX.

MiloradCvjetkovic, 2024-10-22:
Can you try the attached file 
[test_release_driver_l4.MC.zip](https://github.com/user-attachments/files/17473418/test_release_driver_l4.MC.zip)

minicarpet, 2024-10-23:
Hi,
Still same issue, I still receive the wrong data (always 0x11) instead of the ascii code whatever I press.

MiloradCvjetkovic, 2024-10-23:
Hi @minicarpet , 
I unfortunately do not have that board (B-L462E-CELL1), so I can't try your project on the hardware.
I did try to make a project from scratch using your code on the B-L475E-IOT01A and it worked as expected.
- Do you have any other board you can try on?
- Can you enable [debugging events](https://arm-software.github.io/MDK-Middleware/latest/USB/usbd_create_app.html#usbd_debugging) in your project and see if the debug events give some indication about what is going wrong?
- Can you use latest [MDK-Middleware v8.0.0](https://github.com/ARM-software/MDK-Middleware/releases/tag/v8.0.0) which comes with source files of the components thus providing debugging using sources?
This will, however, require some [migration](https://arm-software.github.io/MDK-Middleware/latest/USB/usb_migration.html) from old project.

minicarpet, 2024-10-23:
Hi @MiloradCvjetkovic,

Your project is working probably because your device support OTG USB. Mine does not.

- You can try with any device which does not have USB OTG (such STM32L462RE or STM32L452RE which has [nucleo](https://www.st.com/en/evaluation-tools/nucleo-l452re.html))
- I've attached the EventRecorder.csv file, the interested part are Event 417 were I cannot see any error.
- I actually already use MDS-Middleware v8.0.0 with RTX sources files

MiloradCvjetkovic, 2024-10-23:
Hi @minicarpet , 

as far as I can see NUCLEO-XXXXRX do not have user USB, so these aren't a good fit.
I agree, it might be related that MCU having USB peripheral instead of OTG_FS peripheral.
Where did you attach EventRecorder.csv file?

minicarpet, 2024-10-23:
My bad sorry I forgot,
Here is the file [EventRecorder.csv](https://github.com/user-attachments/files/17490591/EventRecorder.csv)

MiloradCvjetkovic, 2024-10-23:
There is nothing odd in that log.
I'll try to get a device with USB peripheral and try to duplicate the issue.

minicarpet, 2024-11-05:
Hi,
any news about it ?

MiloradCvjetkovic, 2024-11-05:
Hi @minicarpet , 

should be addressed with the commit b10490b07a23d34c057b70b82f2c18c43046b635 .

Can you verify ?

minicarpet, 2024-11-06:
Hi @MiloradCvjetkovic !

CDC class seems ok, thank you:
[image]

But seems that CustomClass has issue - I don't receive any data, I will redo a test later.

minicarpet, 2024-11-07:
Hi again @MiloradCvjetkovic ,

I've tested again this commit. CDC works correctly as I said, but custom class is no more working (at least the reception). I don't receive any IRQ with your modification on my OUT endpoint. I tried to comment the part:
```
// To refresh the PMA configuration the endpoint has to be opened
//if ((req_ep != 0U) && (ptr_ro_info->ptr_rw_info->ep_info[ep_num][EP_OUT_INDEX].configured != 0U)) {
  //(void)HAL_PCD_EP_Open(ptr_ro_info->ptr_hpcd, ep_num, ep_max_packet_size, ep_type);
//}
```

And I can receive data again. This is strange behavior here.
But I got the issue I had before (wrong data are send/receive) even in CDC.
Do you need a test project ? This is more tricky because I have to add libusb code to send and receive data.

MiloradCvjetkovic, 2024-11-07:
Hi @minicarpet , 

let me first try to duplicate the issue myself, so you do not have to send the project for now.

MiloradCvjetkovic, 2024-11-07:
Hi @minicarpet ,

please try with the 8d81e2c8ca30f80dcd4968ba4723631d80b3bbba .

closed 2024-11-26 by minicarpet