Open-CMSIS-Pack/CMSIS-Driver_STM32 #23: [Bug]:  MCI_Get_ClockDivider() return 0 when divider is 1

krma01, 2025-01-31:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

[image]
[image]

 in MCI_STM32_SDMMC.h return 0 when the CLKDIV field is 0.
This result in division by 0 in MCI_STM32.c.

### Expected Behavior

According to reference manual for STM32U585 the divider is 1 when the CLKDIV field is 0. 
The MCI_Get_ClockDivider() should return 1 in this case.

[image]

### Version

1.1.0

### Blocker

- [ ] blocker

VladimirUmek, 2025-02-03:
Hi,

thanks for reporting this bug! I've added commit that addresses the issue - it is referenced above. After a quick check it should be merged to main.

VladimirUmek, 2025-02-20:
The fix has been merged to main branch, hence closing the issue - please reopen on any concerns.

closed 2025-02-20 by VladimirUmek