Open-CMSIS-Pack/CMSIS-Driver_STM32 #25: MX_USB_HANDLE undefined symbol

act-ses, 2025-02-07:
Hi. 
I recently updated our 'old' development environment for the STM32G4 to the latest. (uVision 5.41, CubeMx 6.13.0, STM32G4xx_DFP 2.0.0) 
I migrate an existing project to the new structure and tried to use the cmsis usb driver which comes with the new version. Linker complains about the undefined symbol MX_USB_HANDLE (referred from usbd_stm32.o). 

as far as i can see this handle should be the one defined in usb.c generated by CubeMx. But MX_USB_HANDLE is not generated by CubeMx (MX_Device.h). I also tried a new project but got the same error from the linker.

kind regards

MiloradCvjetkovic, 2025-02-10:
Hi, 

you can check the [migration guide](https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/CubeMX.md#uvision---update-stm32-dfp-packs).

Also, you can check layer for [STM32G474E-EVAL1](https://github.com/Open-CMSIS-Pack/STM32G474E-EVAL1_BSP/tree/main/Layers/Default) board as a reference.

act-ses, 2025-02-10:
Hi,

i already read the linked guide before started the migration. So there were no general issues.
It looks like the MX_Device.h is not propper generated and some information are missing (used the [STM32G474E-EVAL1](https://github.com/Open-CMSIS-Pack/STM32G474E-EVAL1_BSP/tree/main/Layers/Default) board as a reference)

So i added the missing code line in the MX_Device.h file:
```
/* Handle */
#define MX_USB_HANDLE                           hpcd_USB_FS
```
and it work as expected, on the risk being overwritten after the next code generation.
Might be a CubeMX issue rather than CMSIS-Pack.

thanks for pointing me to the reference file.

MiloradCvjetkovic, 2025-02-10:
Hi @act-ses , 

do you, by any chance, have in the CubeMX under `Project Manager -> Code Generator` **selected** checkbox `Generate peripheral initialization as a pair of .c/.h files per peripheral` ?

act-ses, 2025-02-10:
Hi MiloradCvjetkovic,

yes i have selected the checkbox you mentioned as file pair is part of our architecture. I unchecked the box and the MX_Device.h is generated as expected. So no issue with CubeMX. Unfortunately this is not a solution for us as we need the pair of .c/.h files.

However the topic can be closed. Thank you for your support.

closed 2025-02-11 by MiloradCvjetkovic