Open-CMSIS-Pack/CMSIS-Driver_STM32 #26: ETH_MAC_STM32.c incompatible with new HAL drivers (STM32H7)

pukixy, 2025-02-19:
STM has changed the functionality of the HAL_ETH_SetMACFilterConfig function in the new version (which is included in the STM32Cube_FW_H7 V1.12.1 package)

In stm32h7xx_hal_eth.c the changed:
..
((uint32_t)((pFilterConfig->BroadcastFilter == DISABLE) ? 1U : 0U) << 5) |
..
to 
..
((uint32_t)((pFilterConfig->BroadcastFilter == ENABLE) ? 1U : 0U) << 5) |
..

The bit for the broadcast filter means the opposite. No broadcast packets are allowed through. 

If I change this:

  if ((arg & ARM_ETH_MAC_ADDRESS_BROADCAST) != 0U) {  
      eth_mac0_rw_info.mac_filter.BroadcastFilter = DISABLE; //ENABLE;
  } else {
      eth_mac0_rw_info.mac_filter.BroadcastFilter = ENABLE; //DISABLE;
  }

it is working as expected.

MiloradCvjetkovic, 2025-02-19:
Hi @pukixy , 

thank you for raising this issue.

Unfortunately, we are aware of this problem but there is no generic solution.

The solution you proposed would work for new version of H7 HAL, however there are many HALs outside which do not have the same change that you mention.
Also, there are obsolete HALs which will probably never be updated thus will never have the functionality changed.

So, at the moment there is no good solution for this incompatibility.

pukixy, 2025-03-24:
Yes its working. Thanks

MiloradCvjetkovic, 2025-03-25:
Hi @pukixy , 

I will close the issue, feel free to reopen if necessary.

closed 2025-03-25 by MiloradCvjetkovic