Open-CMSIS-Pack/CMSIS-Driver_STM32 #28: [Bug]: STM32H562，USBH_STM32 Compilation error

scsq122, 2025-02-27:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

my environment:
MDK 5.41
CMSIS-Driver_STM32 version 1.1
STM32H562VIT6

my project using USBH_STM32.c, When compiling，some error occur：
compiling USB_Debug.c...
compiling usbh_lib_ehci_tt.c...
compiling usbh_lib_ohci.c...
../Middlewares/CMSIS-Driver/USBH_STM32.c(994): error: no member named 'ep_is_in' in 'USB_HCTypeDef'
  994 |   if (ptr_ro_info->ptr_hhcd->hc[ch].ep_is_in != 0U) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
../Middlewares/CMSIS-Driver/USBH_STM32.c(1188): error: no member named 'ep_is_in' in 'USB_HCTypeDef'
 1188 |   if (ptr_ro_info->ptr_hhcd->hc[ch].ep_is_in != 0U) {
      |       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
../Middlewares/CMSIS-Driver/USBH_STM32.c(1258): error: no member named 'do_ping' in 'USB_HCTypeDef'
 1258 |   ptr_ro_info->ptr_hhcd->hc[ch].do_ping = 0U;
      |   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ^
3 errors generated.
compiling USBH_STM32.c...
compiling USBH_Config_0.c...

### Blocker

- [ ] blocker

MiloradCvjetkovic, 2025-02-28:
Hi @scsq122 , 

thank you for raising this issue.

Can you please try the driver from `usbh_driver` branch at https://github.com/Open-CMSIS-Pack/CMSIS-Driver_STM32/tree/usbh_driver

I do not have that hardware so I cannot try it.

scsq122, 2025-03-01:
@MiloradCvjetkovic I use the new USBH_STM32.c and compiling ok now. thank you。

By the way My hardware works as USB host, I used RTX5 and RL-USB，But u disk is not recognized now。（I make sure the hardware is ok，Because u disk work well in ThreadX）,So I want to know, Is it possible that there is an issue with USBH_STM32? Or maybe other code issue...

MiloradCvjetkovic, 2025-03-03:
Hi @scsq122 , 

it is very much possible that the driver does not work because that DRD variant was not tested on hardware.

I will try to find some hardware that has the same DRD controller and try it there.

MiloradCvjetkovic, 2025-07-10:
Hi @scsq122 ,

support for DRD controller was just added, can you try it?

MiloradCvjetkovic, 2025-08-21:
Hi @scsq122 , 

can you please try with a691b3bf9abab8103f1621cf16210be7a8fb2eda ?

MiloradCvjetkovic, 2025-09-01:
Hi @scsq122 , thanks for confirmation.

I'll close this issue then, and if you stumble upon anything please re-open it or create a new one.

Thank you.

closed 2025-09-01 by MiloradCvjetkovic