Open-CMSIS-Pack/CMSIS-Driver_STM32 #29: PCD Handle error

JonasDo27, 2025-03-19:
I am trying to use the CMSIS Drivers 1.1.0 for a STM32U585 in order to use the Keil MDK Middleware 8.0.0 for an USB CDC Device.
I am using Visual Studio Code as an environment.
I´ve followed the instructions on setting up CubeMX and generating the code.
The following error occurs during compiling:

"Error: L6218E: Undefined symbol MX_USB_OTG_FS_HANDLE (referred from USBD_STM32.o)."

The symbol is declared as extern at USBD_STM32.c:

`extern  PCD_HandleTypeDef       MX_USBD##n##_HANDLE;`

But it is not defined anywhere else in the project.

Am I getting something wrong here?

Thank you

DavidLesnjak, 2025-03-19:
Hi @JonasDo27,

Please check configuration in STM32CubeMX. Did you selected `Mode` = `Device Only` for `USB_OTG_FS` peripheral?
Is MX_Device.h file generated? Is MX_USB_OTG_FS defined in MX_Device.h?

JonasDo27, 2025-03-25:
Hi David,
Yes, mode is set to Device Only and the MX_Device.h is generated with the following section for USB:

`/*------------------------------ USB_OTG_FS     -----------------------------*/
#define MX_USB_OTG_FS                           1

/* Virtual mode */
#define MX_USB_OTG_FS_VM                        Device_Only
#define MX_USB_OTG_FS_Device_Only               1`

DavidLesnjak, 2025-03-25:
Hi @JonasDo27,

It looks like the `MX_USB_OTG_FS_HANDLE` define is missing in your generated `MX_Device.h` file.

I did a quick test on my end using the `STM32U585AIIx` device, and in my case, the `MX_USB_OTG_FS_HANDLE` define was correctly generated. I've attached a screenshot (STM32U585AIIx_USBD.png) for reference.

Could you let me know which version Toolbox you're using?

[image]

DavidLesnjak, 2025-03-25:
Yes, this is currently the expected behavior.

At the moment, our cbridge tool (part of Toolbox) does not support the `Generate peripheral initialization as a pair of .c/.h files per peripheral` option in STM32CubeMX.

We’re actively working on adding support for this mode, and it’s expected to be included in CMSIS-Toolbox version 2.9.0.

JonasDo27, 2025-03-25:
Alright, thank you!