Open-CMSIS-Pack/CMSIS-Driver_STM32 #32: [Bug]: MCI Driver error: MCI_STM32.c(333): error: initializer element is not a compile-time constant

rossb93, 2025-07-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

When trying to building the project with with MCI1 enabled using MCI driver I get an error:

CMSIS-Driver_STM32\1.2.0\Drivers\MCI_STM32.c(333): error: initializer element is not a compile-time constant
  MCI1_RxDma,
  ^~~~~~~~~~
1 error generated.

I'm using ARM compiler v6.18

### Expected Behavior

Build without issue

### Steps To Reproduce

1. Keil project with CubeMX generator
2. Enabling SDMMC1 with DMA RX and TX
3. In Keil enabling CMSIS Driver -> MCI software component 
4. Build project

### Version

CMSIS Driver STM32: 1.2.0
MCI 2.4.0

### Blocker

- [x] blocker

jkrech, 2025-07-17:
Which version of CubeMX and STM32Cube Firmware Library are you using?
Which version of the CMSIS-Toolbox are you using?

HansSchneebauer, 2025-07-17:
Very good. Do you agree to close this issue?

closed 2025-07-17 by jkrech