Open-CMSIS-Pack/CMSIS-Driver_STM32 #33: [Bug]: Missing #ifdef I2C_VARIANT_HAS_FILTER around anf_en call in I2C_STM32.c

rossb93, 2025-07-17:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

See the Error: no member named 'anf_en' in 'RO_Info_t'
if (ptr_ro_info->anf_en != 0U) {
~~~~~~~~~~~  ^

On my board I2C_VARIANT_HAS_FILTER is not defined so the anf_en member gets compiled out, so its usage in the line above (line 1428 of file I2C_STM32) should also be compiled out

### Expected Behavior

No error.

### Steps To Reproduce

Build with I2C_STM32 included in project with 

### Version

ARM CMSIS Driver 2.10.0
I2C (API) -> I2C: Version 3.0.0

### Blocker

- [ ] blocker

andbararm, 2025-07-18:
Ok, I had a look. Given that struct RO_Info_t has conditional compilation for this:

```typedef struct {
  I2C_HandleTypeDef            *ptr_hi2c;               // Pointer to I2C handle
  RW_Info_t                    *ptr_rw_info;            // Pointer to run-time information (RW)
  uint32_t                      peri_clock_freq;        // Peripheral clock frequency (in Hz)
  PinConfig_t                   scl_pin_config;         // SCL pin configuration structure
  PinConfig_t                   sda_pin_config;         // SDA pin configuration structure
#ifdef I2C_VARIANT_HAS_FILTER
  uint16_t                      anf_en;                 // Analog noise filter enable
  uint16_t                      dnf;                    // Digital noise filter coefficient value (0 - disabled)
#endif
} RO_Info_t;
```
locations where the "anf_en" element is used should have the same.

However, I think there is a bit more to this. Given that CubeMX defines in MX_Device.h:

```/* Filter Settings */
#define MX_I2C1_ANF_ENABLE                      0
#define MX_I2C1_DNF                             0
```

the compilation check to define I2C_VARIANT_HAS_FILTER or not does not work:

```#if    (defined(MX_I2C1_ANF_ENABLE) || defined(MX_I2C1_DNF) || \
        defined(MX_I2C2_ANF_ENABLE) || defined(MX_I2C2_DNF) || \
        defined(MX_I2C3_ANF_ENABLE) || defined(MX_I2C3_DNF) || \
        defined(MX_I2C4_ANF_ENABLE) || defined(MX_I2C4_DNF) || \
        defined(MX_I2C5_ANF_ENABLE) || defined(MX_I2C5_DNF) || \
        defined(MX_I2C6_ANF_ENABLE) || defined(MX_I2C6_DNF) || \
        defined(MX_I2C7_ANF_ENABLE) || defined(MX_I2C7_DNF) || \
        defined(MX_I2C8_ANF_ENABLE) || defined(MX_I2C8_DNF))
#define I2C_VARIANT_HAS_FILTER          1
#endif
```

Have you modified this already yourself?  Else, I would say  I2C_VARIANT_HAS_FILTER is always defined as 1. The check should be something like this:

```#if    ((defined(MX_I2C1_ANF_ENABLE)&&(1==MX_I2C1_ANF_ENABLE)) || (defined(MX_I2C1_DNF)&&(1==MX_I2C1_DNF)) || \```

rossb93, 2025-07-18:
Hm so I realised that CubeMX doesn't seem to be generating these filter settings whether I have the analog filter enabled or not. If I add them manually into MX_Device.h the error goes away since they have now been defined

MiloradCvjetkovic, 2025-07-18:
Which version of cmsis_toolbox are you using?

MiloradCvjetkovic, 2025-07-18:
Can you update cmsis_toolbox to latest (v2.10.0) and try?

MiloradCvjetkovic, 2025-07-18:
Can you then please close the issue if it is actually not an issue @rossb93 ?

rossb93, 2025-07-18:
I think there is still technically a bug in the code where the filter setting definitions are not defined in the MX_Device.h file it doesnt compile out the usage of anf_en correctly but I can close this anyway since that only happens due to a bug in the old toolbox version.

MiloradCvjetkovic, 2025-09-15:
Hi @rossb93 , 

I'm closing this issue since it was addressed and fix is available in latest released version of pack.

closed 2025-09-15 by MiloradCvjetkovic